Team-2502/RobotCode2018 #29: New LIDAR Sensor

UltraX3, 2018-01-14:
Mentor _Chris Olson_ had an idea that we should look into or try LIDAR sensors (which I previously thought would cost mucho dinero). I have not looked deeply into this, but it sounds interesting — especially if the cheap LIDAR sensors are any good.

Note: to get data from cheaper LIDAR sensors (at least from some that I saw) we might need to use an Arduino and Arduino C and then probably transfer it through Network Tables. I am not sure whether this is practical or not.

isaacaj, 2018-01-14:
I think that a LIDAR sensor would present a great learning experience for us. We would most likely not end up doing anything with it until after build season, though. 

A cool long term or summer project could involve some sort of collision detection/avoidance.

UltraX3, 2018-01-21:
Randall Maas talked with me about this today. He thought LIDAR sensors are pretty useful and can be used with triangulation (seeing distance between 3 known points) to determine the absolute location of the robot.

UltraX3, 2018-02-11:
@donovanMaas

andrewgazelka, 2018-03-21:
Is not worth trying to add for this current robot. Will be considered for the summer.

closed 2018-03-21 by andrewgazelka