Team-2502/RobotCode2018 #39: PID tuning does not work for left motors

UltraX3, 2018-01-21:
PID tuning does not work for left drive train motors (it does for right). This is almost surely because the left talons are reversed. Even when the left motors are set to inverted, the proportional error still grows when getting closer and closer to the target velocity. This makes it seem like talon is thinking the motor is getting further (going in the opposite direction) from desired when actually getting closer.

ritikmishra, 2018-01-21:
On herschel, one side still uses a quadrature encoder, whilst the other uses CTRE mag encoders. I do not remember which, but you can take a peek when the robot bottom is visible while it is on the cart.

UltraX3, 2018-01-21:
As of a ~week ago both now use CTRE mag encoders :)

UltraX3, 2018-01-21:
This needs to work soon for pure pursuit since `voltage ratios != wheel speed ratios`, and pure pursuit relies on these ratios to function.

UltraX3, 2018-01-22:
Update: Michael thinks he inverted the right talons, but it seems like both sides are now inverted. Can some programmer verify this?

UltraX3, 2018-01-22:
Fixed after @MDW01 changed some things w/ electronics.

closed 2018-01-22 by UltraX3