Team-2502/RobotCode2018 #47: Be able to set waypoint headings and velocities

UltraX3, 2018-01-23:


ritikmishra, 2018-01-23:
I think that the angle can be best achieved with the use of intermediate waypoints

UltraX3, 2018-01-23:
@ritikmishra I agree that intermediate waypoints should _help_ determine heading, and maybe a modified Pure Pursuit would determine velocity at a certain period. However, when picking up cubes I am thinking we will need a specific heading (and lowered velocity). Maybe we can combine Pure Pursuit with some more basic trajectory alg when we get very close to the destination?

UltraX3, 2018-01-23:
what probably makes more sense is being able to set max velocities between waypoints

ritikmishra, 2018-03-15:
Perhaps we can fix theta in splines where we can set the derivative at a control point
(speed works)

andrewgazelka, 2018-03-21:
If this were added, it would be with splines.

ritikmishra, 2018-05-18:
Alright, I've figured out how to make parametric cubic splines. Hopefully this will be fixed soon

andrewgazelka, 2018-05-18:
Yay. 👍 let's close this after we do unit tests in the simulator.

ritikmishra, 2018-05-20:
Unit tests on splines have been completed in the simulator and are quite successful.

🎉

closed 2018-05-20 by ritikmishra