Team-2502/RobotCode2018 #52: Revert to Stable Functionality

andrewgazelka, 2018-02-16:
# Revert Functionality
## Idea
If a programmer makes a change at the competition which breaks the robot, the driver will be able to press a button to revert back to a more stable version of the robot code.

ritikmishra, 2018-02-16:
I believe GitKraken offers this functionality.

andrewgazelka, 2018-02-16:
@ritikmishra I am referring to changing functionality _once the match has started_. I have heard from mentors that there are often arguments between programmers and drivers when programmers make last-minute (untested) changes that will
> totally work

Of course, the drivers are usually not this confident. By implementing this system, programmers could gain more trust from drivers by letting them know that even if things go horribly wrong, they will still be able to revert to a working version with a press of a button.

andrewgazelka, 2018-03-21:
probably not useful enough to include... would only be useful in autonomous and thus couldn't have human input.

closed 2018-03-21 by andrewgazelka