Team-2502/RobotCode2018 #58: Motion profiling elevator

andrewgazelka, 2018-02-20:
https://www.chiefdelphi.com/forums/showpost.php?p=1730497&postcount=5

ritikmishra, 2018-03-15:
Almost completely implemented in `ae9143`. Just need to uncomment some things upon confirmation that the refactoring didn't break anything.

andrewgazelka, 2018-05-18:
To do this reliably we will need planetary gear sliced encoders (I think they are called lol).

Closing because we are not doing it on this robot.

closed 2018-05-18 by andrewgazelka

ritikmishra, 2018-05-18:
> To do this reliably we will need planetary gear sliced encoders (I think they are called lol).
Wrong. Our encoder is actually quite reliable as long as the shaft is all the way in (a plate now exists to prevent shaft from coming out)

The thing stopping this is how mechanically jerky the elevator is. Preferably, we would be able to just run the minicims to run the elevator. Unfortunately, we must do the lock engage/disengage thing, and even then, the minicims must be running at a minimum speed in order to not stall out. To fix this, we would need to counterbalance the elevator with some sort of spring.