Team-2502/RobotCode2018 #64: SQLite PP record

andrewgazelka, 2018-03-21:
_qtd. from Slack_

Do you think it would be logical to use SQLite on the driver station for "recording" of autonomous?

Pros: 
- easier to do queries
- we won't need to do code for custom file format
- all matches could easily be transferred with a USB via one SQLite file (doesn't involve connecting to Ethernet, which waists precious time)

Cons:
- Will require a separate application running on driver station which will read from network tables. We will need to figure out how to do this... possibly make a custom Shuffleboard widget with no graphics but functionality we need? Also I think 254 made a custom NetworkTables reader in Python... probably fairly easy to get pertinent info on the network.

ritikmishra, 2018-05-18:
@andrewgazelka @MDW01 Do we still want to make this in SQLIte? The infrastructure already exists for doing it in a CSV . . .

ritikmishra, 2018-05-18:
Plus the roboRIO has a WebDAV server so it should be trivial to drag+drop the CSV on the RIO to a computer for visualization. Finder on our school macs will let us connect to WebDAV servers.

andrewgazelka, 2018-05-18:
@ritikmishra CSV is the way to go, then.

closed 2018-05-18 by andrewgazelka