Team-2502/RobotCode2018 #70: Pause before starting the autonomous

ritikmishra, 2018-04-05:
This issue is a pretty obvious one and I'm surprised that this hasn't been added yet

Look at literally any match of ours on TBA and watch our robot

> Emcee: 3 . . 2 . . 1 . . Power Up!
> Our robot: _does nothing_
> Our robot after around 1.5 seconds: _autonomous_

This is a strange issue and it is likely caused by something pure pursuit is initializing. The profiler (https://www.chiefdelphi.com/forums/showthread.php?t=160071) should show where our process is hanging.

andrewgazelka, 2018-04-05:
Also, a high probability it was `MathUtils`. 

_Even though I walk through the valley of the shadow of unoptimized software, I will fear no pauses, for YourKit is with me_.

andrewgazelka, 2018-05-18:
Did we ever figure out if this was fixed? @ritikmishra

ritikmishra, 2018-05-22:
I don't think so because we never ran `develop` auton in a "production" environment